SiLab-Bonn/irrad_control #21: Add offline analysis and plot generation

leloup314, 2019-11-29:
Enable offline plotting / analyzing of the output files generated during irradiation

DnnsSrlnd, 2020-03-20:
Automated Analysis of Current-Calibration Data for the SEM:

After measurements with an external faraday cup (FC) the measured FC current is correlated to the sum of the signals of all four SEM-foils. This enables to obtain a current calibration for an detector irradiation when an external faraday cup is not present.

The calibration data should automatically be analyzed and following features should be usable:

1. Conversion of binary .h5 files into tabular separated human readable datafiles (.dat) with or without subtracting the measurement offsets.

2. Generation of a histogram of I_cup / U_sem of the measurement. Here, I_cup is the FC current, converted from the known full-scale-voltage (5 Volt corresponds to e.g. 1000 nA). U_sem is the converted voltage signal coming from the current-sum of all SEM-foils. Average and standard deviation should be extracted from the histogram if the resulting distribution is gaussian shaped.

3. Plotting the data U_sem vs I_sem in a heatmap and fitting a linear function on them while using several filters on the collected data. 
Filter 1: The beam should pass the center of the SEM during calibration. Therefore data points should be omitted which have a relative position (vertical plane (U-D)/(U+D) and horizontal plane (L-R)/(L+R) where U, D, L and R are the offset-subtracted upper, lower, left and right SEM-foils, respectively) which is a certain distance away from the SEM center.
Filter 2: Outliers which show a certain I_cup / U_cup which is a certain distance (in reference of standard deviations) too far away from the average I_cup / U_cup should be omitted.
Out of the slope of the linear fit, the proportionality M between U_cup and the beam current should be extracted (with its error). 
Using the relation Y= (Q_beam/Q_electron) FSV / (5V  M) the secondary electron yield (SEY) Y should be extracted. Here, Q_beam is the electric charge of the beam particle and Q_electron is the charge of the electron. FSV is the full-scale-current of the SEM readout electronics.

In a later version one could also think about an automated import of the calibration data into a calibration database where also interpolations between different energies for all available particles were conducted. This would reduce the SEM-calibration-factor choosing to the question of type and energy of the beam particles.

leloup314, 2020-03-23:
I agree! The most commonly needed analysis steps (beam current calibration, secondary electron yield, hardness factor calibration, etc) should be implemented as part of this package. I suggest we also implement plotting, but separately. Let's make a feature branch for this, so we can work on this together!

leloup314, 2020-03-24:
I made a feature branch for this (see [here](https://github.com/SiLab-Bonn/irrad_control/tree/offline_analysis/irrad_control)). @DnnsSrlnd you can pull and push to this branch directly

leloup314, 2022-07-27:
Now in `development` branch with 7b16fa9e0559f69d6f3a6b5e839b11f6b0b6d7e6

closed 2022-07-27 by leloup314