SiLab-Bonn/irrad_control #24: Automated Calculation of Time-Average and Standard-Deviation within Timeframe

DnnsSrlnd, 2020-02-27:
An automated calculation of the time-average and standard deviation of all selected channels within the displayed timeframe would be nice.

leloup314, 2020-02-27:
This is about to go into the development branch, thanks :+1: Once it's in master I will close this

leloup314, 2020-03-24:
@DnnsSrlnd do I remember correctly that the graphs in the scrolling data plots are now to thin for good visuals?

DnnsSrlnd, 2020-03-24:
Yes, their are either too thin or now have a transparent feature but I would suspect the first.

leloup314, 2020-03-24:
It's a mixture of both ;) I will change this

leloup314, 2020-03-24:
I changed the behavior of the stats "window". See df4a908ddf67746f57cf716d27893321e32a77a6. Should be better now

leloup314, 2020-04-06:
Added statistics for all plots in 814d288aa8275d8de1a068185c93fdb4e9574ef4

leloup314, 2020-07-05:
Close in favor of #35

closed 2020-07-05 by leloup314