SiLab-Bonn/irrad_control #29: Error due to identically labeled channels

DnnsSrlnd, 2020-06-03:
irrad_control does not seem to be able to close itself "elegantly" when you name two (temperatur) channels with identical lables.

It this case, two temperature sensors on a connected Arduino were accidentally named "t1" which resulted in the following bash output:

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/home/irradiation/miniconda2/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/home/irradiation/miniconda2/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/irradiation/git/irrad_control/irrad_control/utils/daq_proc.py", line 356, in recv_cmd
    self.handle_cmd(**cmd_dict)
  File "/home/irradiation/git/irrad_control/irrad_control/converter.py", line 480, in handle_cmd
    self._start_interpreter(data)
  File "/home/irradiation/git/irrad_control/irrad_control/converter.py", line 467, in _start_interpreter
    self._setup_daq()
  File "/home/irradiation/git/irrad_control/irrad_control/converter.py", line 178, in _setup_daq
    self.temp_data[server] = np.zeros(shape=1, dtype=temp_dtype)
ValueError: field 't1' occurs more than once

leloup314, 2020-06-05:
Thanks. Generally, everything that goes into `numpy` s structured arrays needs to have an unique identifier for the lookup. Otherwise `numpy` will raise `ValueError: two fields with the same name`

leloup314, 2020-07-05:
Close in favor of #34

closed 2020-07-05 by leloup314