SiLab-Bonn/irrad_control #45: Clean-up of `rad_monitor` branch

leloup314, 2021-02-23:
@christian235 please clean up the redundant files in the `rad_monitor` branch so the folder structure looks like this:

- rad_monitor:
    -> `__init__.py` 
    -> iseg_hv.py
    -> arduino_freq_counter.py
    
     - ArduinoFreqCount: 
         -> ArduinoFreqCounter.ino

        - FreqCounter:
            -> `The files of the FreqCounter library` (utils, examples, ...)

Also, make sure to test and comment ;)

christian235, 2021-02-23:
I have a folder called  "__pycache__" (__ __). Normally I would delte this one but in this case with the double underscore I'm kind of confused. Can I delete this kind of backup or is it necessary for anything?

leloup314, 2021-02-23:
You can delete it, simply ignore it or explicitly add it to the `.gitignore`-file in order to have `git` ignore it. More info [here](https://stackoverflow.com/questions/16869024/what-is-pycache)

christian235, 2021-02-24:
Finally i tested all functions and now everything is working correct. So you can start to include rad_monitor.

leloup314, 2021-02-24:
Looks fine. Can you still add the missing FreqCount library files into the `ArduinoFreqCount` folder?

christian235, 2021-02-24:
Sorry for the merge. I forgot that Ihad previous changes. Im not sure if we need the folder 'advance' but just for the case I also included this one.

leloup314, 2021-02-24:
I just pushed what I wanted. Now it looks like in the description above

leloup314, 2022-03-31:
Closed with #49

closed 2022-03-31 by leloup314