SiLab-Bonn/irrad_control #51: Basil driver

DavidLP, 2021-06-22:
It would be awesome to have a basil driver, to allow synchronization between chips measurements with the fluence. Some simple interface, like start/pause/stop/get_fluence.

DavidLP, 2021-06-22:
get_temperature/get_humidity

DavidLP, 2021-06-22:
get_status (ERROR/OK/RUNNING/PAUSE)

leloup314, 2021-06-22:
I don't fully understand what you mean? To be run during irradiation?

leloup314, 2021-06-22:
What underlying structure are you thinking? Retrieving the data directly from the irrad site DAQ PC via network?

DavidLP, 2021-06-22:
> I don't fully understand what you mean? To be run during irradiation?

Yes. Would be awesome to check TID with TID sensor of RD53 every 10th scan or so.
Then we can check influence of biasing / clock to chip and without bias. This would be a really
proper measurement of interest.

> What underlying structure are you thinking? Retrieving the data directly from the irrad site DAQ PC via network?

Yes. ZMQ over Ethernet? Basil devices have init/close procedures; there one can setup the connection. Can show you.

DavidLP, 2021-06-22:
Generally, it is of interest to check chip parameters during irradiation. Such a basil driver would simplify this.